sinara-hw/KASLI-DIOT #2: SCH and PCB review

gkasprow, 2024-03-22:
- [ ] add bootloader button (https://github.com/sinara-hw/Kasli/issues/107)

filipswit, 2024-03-24:
> Move the bootloader button closer to the edge

I don't think we can move it closer. It can't fit under the panel. We can optionally move it on TOP.

> Mark 12V connector as DNP or move it to the back - now it obstructs the airflow

We can try to move it to the right. For the first proto it will be needed to work on the table, later might be unmounted.
[image]

gkasprow, 2024-03-24:
Nie musisz odrazu pprawiac issues. Pawel bedzie robil review i pewnie i tak
cos wyjdzie.

filipswit, 2024-11-10:
12V Connector moved, in future can be marked as DNP
[image]

closed 2024-11-10 by filipswit