sinara-hw/KASLI-DIOT #36: DDR routing

marmeladapk, 2024-07-02:
Highlighted lines are too close. Saturn says that over 1 cm there could be 0,5V of crosstalk. You can use bottom layer to route 3 signals, Altium calculates via length but it won't take into account the propagation delay difference between different layers, though it's pretty minor difference (6.23 ps/mm vs 6.86 ps/mm). Alternatively if you use matched length with delay values it won't calculate via delay :/

[image]

There's also 2 lines like that in BYTE2.

marmeladapk, 2024-07-03:
Also there are some lines on layer 5 and 6 that run parallel to each other over long distances and don't have a plane inbetween. Maybe move them around a bit?

filipswit, 2024-11-10:
Made bigger separation on L6 wherever was possible, and moved traces ln L5 and L6 so they mostly don't run in parallel.

closed 2024-11-10 by filipswit