sinara-hw/KASLI-DIOT #42: SERVMOD control

gkasprow, 2024-08-31:
At the moment, SERVMOD is controlled using FPGA GPIOs
This means I2C resources of the DIOT crate wouldn't be available from USB I2C.
I'd connect them (maybe in parallel) to I2C extender.

gkasprow, 2024-09-03:
The same applies to RST_N; we need to make sure we can enable power supplies of peripheral boards
The RST_N must be low by default before FPGA is configured.
It needs to be strong enough to hold pullups of 8 modules

gkasprow, 2024-11-05:
Since existing I2C expanders cannot properly steer a 200-ohm load in DIOT devices, we need to add something that buffers them.
Let's use existing (in the BOM) 16-bit expanders followed by LVC245, which provides 24mA of current capability. At a 3.3V supply, it will deliver more than 2.4V at such a load.
Please connect it in the following way:
Expander Port 0 -> 1k series resistors ->HC245 -> 47k pullups-> SERVMODs
Expander Port 1 -> HC245 OE, DIR. 
This way, we would be able to read which SERVMODs are pulled by 200Ohm to GND and then drive them high to enable configuration readout.
Expander Port 1 should also drive RST_N
Keep existing connections with FPGA via DNP resistors

filipswit, 2024-11-10:
Added, with default PCA9539 
[image]

gkasprow, 2024-11-10:
Are you sure HC245 has enough current capability to steer 200R to the proper logic level?

closed 2024-11-11 by filipswit